Saiyato/volumio-rotary-encoder-plugin #1: Volumio GUI freezes after plugin setup

mundodisco8, 2018-01-26:
Hi,

I installed your plugin (latest version) on a fresh installation of the latest version of volumio (grabbed today from the web page).

As soon as I save the configuration, I get thrown to the reboot screen (the animation of rotating dots). The web GUI is not accessible anymore, but volumio replies to pings, so it's working. I tried rebooting, but I can't get to the GUI. I tried reinstalling to reproduce the problem, and the same thing happened.

I'm using a bourns encoder (http://www.bourns.com/docs/Product-Datasheets/PEC11R.pdf?sfvrsn=bb617cbf_5) and the schematic for my circuit is the following:

[image]

I'm using a raspberry pi 2 with a HifiBerry Amp+, and GPIO 22 for A and 17 for B and 4 for the switch. I've managed to use the rotary on raspbian, when I did some tests, and it worked there.

Any ideas? Is there a log I can attach to help you troubleshoot?

Cheers!

Saiyato, 2018-01-26:
Hi,

Thanks for the complete feedback, I think something's off in saving the config. I did test this, but when pushing to production, bugs are bound to be found.

Can you run a 

> journalctl -f

 while saving your config? I'm expecting an error to be thrown, once you see an error you can ctrl + c (abort) the trace, I'm going to need the stacktrace of the error, which is found near a ^ sign.

I hope I'm clear enough, otherwise we'll get there in a few tries. ;)

josonxie, 2018-01-27:
volumio@volumio:~$ volumio volume
curl: (7) Failed to connect to 127.0.0.1 port 3000: Connection refused

josonxie, 2018-01-27:
I also have the same problem

volumio@volumio:~$ volumio volume
curl: (7) Failed to connect to 127.0.0.1 port 3000: Connection refused

Can not enter the web ui

mundodisco8, 2018-01-28:
Hi, 

not sure if I did the right thing, but I was not getting any logs with journalctl -f so I followed [these instructions](https://unix.stackexchange.com/questions/159221/how-do-i-display-log-messages-from-previous-boots-under-centos-7) to get it working. After a reboot, I got this. After activating the plugin I would get the same warning messages, with a different, increasing number each time:
```
-- Logs begin at Sun 2018-01-28 11:05:41 UTC. --
Jan 28 11:07:35 volumio systemd[1766]: Reached target Paths.
Jan 28 11:07:35 volumio systemd[1766]: Starting Timers.
Jan 28 11:07:35 volumio systemd[1766]: Reached target Timers.
Jan 28 11:07:35 volumio systemd[1766]: Starting Sockets.
Jan 28 11:07:35 volumio systemd[1766]: Reached target Sockets.
Jan 28 11:07:35 volumio systemd[1766]: Starting Basic System.
Jan 28 11:07:35 volumio systemd[1766]: Reached target Basic System.
Jan 28 11:07:35 volumio systemd[1766]: Starting Default.
Jan 28 11:07:35 volumio systemd[1766]: Reached target Default.
Jan 28 11:07:35 volumio systemd[1766]: Startup finished in 52ms.
Jan 28 11:10:17 volumio node[2862]: *** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
Jan 28 11:10:17 volumio node[2862]: *** WARNING *** Please fix your application to use the native API of Avahi!
Jan 28 11:10:17 volumio node[2862]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
Jan 28 11:10:17 volumio node[2862]: *** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the Apple Bonjour compatibility layer of Avahi.
Jan 28 11:10:17 volumio node[2862]: *** WARNING *** Please fix your application to use the native API of Avahi!
Jan 28 11:10:17 volumio node[2862]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>
Jan 28 11:10:29 volumio node[2927]: *** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
Jan 28 11:10:29 volumio node[2927]: *** WARNING *** Please fix your application to use the native API of Avahi!
Jan 28 11:10:29 volumio node[2927]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
Jan 28 11:10:29 volumio node[2927]: *** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the Apple Bonjour compatibility layer of Avahi.
Jan 28 11:10:29 volumio node[2927]: *** WARNING *** Please fix your application to use the native API of Avahi!
Jan 28 11:10:29 volumio node[2927]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>
```

After a reboot, a similar situation
```
-- Logs begin at Sun 2018-01-28 11:13:04 UTC. --
Jan 28 11:15:55 volumio systemd[1845]: Reached target Paths.
Jan 28 11:15:55 volumio systemd[1845]: Starting Timers.
Jan 28 11:15:55 volumio systemd[1845]: Reached target Timers.
Jan 28 11:15:55 volumio systemd[1845]: Starting Sockets.
Jan 28 11:15:55 volumio systemd[1845]: Reached target Sockets.
Jan 28 11:15:55 volumio systemd[1845]: Starting Basic System.
Jan 28 11:15:55 volumio systemd[1845]: Reached target Basic System.
Jan 28 11:15:55 volumio systemd[1845]: Starting Default.
Jan 28 11:15:55 volumio systemd[1845]: Reached target Default.
Jan 28 11:15:55 volumio systemd[1845]: Startup finished in 66ms.
Jan 28 11:16:00 volumio node[1828]: *** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
Jan 28 11:16:00 volumio node[1828]: *** WARNING *** Please fix your application to use the native API of Avahi!
Jan 28 11:16:00 volumio node[1828]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
Jan 28 11:16:00 volumio node[1828]: *** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the Apple Bonjour compatibility layer of Avahi.
Jan 28 11:16:00 volumio node[1828]: *** WARNING *** Please fix your application to use the native API of Avahi!
Jan 28 11:16:00 volumio node[1828]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>
Jan 28 11:16:13 volumio node[1908]: *** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
Jan 28 11:16:13 volumio node[1908]: *** WARNING *** Please fix your application to use the native API of Avahi!
Jan 28 11:16:13 volumio node[1908]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
Jan 28 11:16:13 volumio node[1908]: *** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the Apple Bonjour compatibility layer of Avahi.
Jan 28 11:16:13 volumio node[1908]: *** WARNING *** Please fix your application to use the native API of Avahi!
Jan 28 11:16:13 volumio node[1908]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>
```

I recorded a screencast of the installation and activation:
https://www.screencast.com/t/HiUDBzlR

I don't think this is the info you were looking for. Give me some pointers on how to get it if it isn't.

Curry444, 2018-01-28:
I have the same problem. As soon as I save the configuration, I get thrown to the reboot screen (the animation of rotating dots). The web GUI is not accessible anymore.
Raspberry PI3 , latest fresh volumio installation without other plugins and a iqaudio dac plus audio card/amplifier.   Im a noob in linux, so i cant give you more details.

Greetings 
Dennis

Saiyato, 2018-01-29:
I will try to investigate this week.

Saiyato, 2018-01-29:
Found it! The configuration is not written!

Saiyato, 2018-01-29:
```
Jan 29 10:05:37 volumiolcd volumio[7174]: info: PLUGIN START: rotaryencoder
Jan 29 10:05:37 volumiolcd volumio[7174]: info: Could not initiate rotary encoder #1 with error: Error: EINVAL: invalid argument, write
Jan 29 10:05:37 volumiolcd volumio[7174]: /data/plugins/user_interface/rotaryencoder/index.js:269
Jan 29 10:05:37 volumiolcd volumio[7174]: self.firstEncoder.on('rotation', direction => {
Jan 29 10:05:37 volumiolcd volumio[7174]: ^
Jan 29 10:05:37 volumiolcd volumio[7174]: TypeError: Cannot read property 'on' of undefined
Jan 29 10:05:37 volumiolcd volumio[7174]: at rotaryencoder.constructFirstEncoder (/data/plugins/user_interface/rotaryencoder/index.js:269:19)
Jan 29 10:05:37 volumiolcd volumio[7174]: at rotaryencoder.onStart (/data/plugins/user_interface/rotaryencoder/index.js:42:8)
Jan 29 10:05:37 volumiolcd volumio[7174]: at PluginManager.startPlugin (/volumio/app/pluginmanager.js:256:27)
Jan 29 10:05:37 volumiolcd volumio[7174]: at HashMap.<anonymous> (/volumio/app/pluginmanager.js:332:29)
Jan 29 10:05:37 volumiolcd volumio[7174]: at HashMap.forEach (/volumio/node_modules/hashmap/hashmap.js:157:10)
Jan 29 10:05:37 volumiolcd volumio[7174]: at HashMap.proto.(anonymous function) [as forEach] (/volumio/node_modules/hashmap/hashmap.js:169:7)
Jan 29 10:05:37 volumiolcd volumio[7174]: at PluginManager.startPlugins (/volumio/app/pluginmanager.js:331:15)
Jan 29 10:05:37 volumiolcd volumio[7174]: at new CoreCommandRouter (/volumio/app/index.js:63:24)
Jan 29 10:05:37 volumiolcd volumio[7174]: at Object.<anonymous> (/volumio/index.js:34:21)
Jan 29 10:05:37 volumiolcd volumio[7174]: at Module._compile (module.js:570:32)
Jan 29 10:05:37 volumiolcd volumio[7174]: at Object.Module._extensions..js (module.js:579:10)
Jan 29 10:05:37 volumiolcd volumio[7174]: at Module.load (module.js:487:32)
Jan 29 10:05:37 volumiolcd volumio[7174]: at tryModuleLoad (module.js:446:12)
Jan 29 10:05:37 volumiolcd volumio[7174]: at Function.Module._load (module.js:438:3)
Jan 29 10:05:37 volumiolcd volumio[7174]: at Module.runMain (module.js:604:10)
Jan 29 10:05:37 volumiolcd volumio[7174]: at run (bootstrap_node.js:389:7)

```

closed 2018-01-29 by Saiyato

Saiyato, 2018-01-29:
The problem was the template used to create the plugin, I've raised questions about this on the forum.

mundodisco8, 2018-02-05:
Hey! I tried the latest version and it seems to fix the problem, good job!

commit 70ca1ceb (2018-01-29): Major update

- Removed unnecessary modules
- Fixed the config issue

fixes #1