Saiyato/volumio-rotary-encoder-plugin #4: Hw Debouncing

Hyldig01, 2018-03-08:
Hi, 

Im working on setting up a rotary encoder for my volumio setup, 
for now the volume is bouncing all around when i use it.  so i am trying to fix that. 
I ordered some Ceramic Disc Capacitors 50V 100nF 0.1uF 104pF to connect between the GND and CLK and DT... However i am not to sure about electronics, so i would just like to check if that is all i need for Hardware Debouncing?

Hyldig01, 2018-03-12:
so the Ceramic Disc Capacitors 50V 100nF 0.1uF 104pF landed today, and a quick solder later, it almost worked perfectly... at least volume up worked, but still a little buggy with volume down... 
hopeing for some inputs to make it work flawlessly :D

Saiyato, 2018-03-12:
Hi,

I'm not an electro-engineer, I just took the advise I found online ;) so unfortunately I can't really help you with that. I know that debouncing literally is a bounce of the contact and a capacitor prevents this by "eating" the slight extra contact (before or) after the real "press".

Not sure what you mean by buggy, but the commit from diehardsk works perfectly for me. I still need to wait for Volumio to accept my commits, but if you download the zip-file it should work great.

Hyldig01, 2018-03-12:
it still jump a little, but mostly when i turn down the volume... there it sometimes goes up instead :/ 

Thanks for all the great work by the way :D 
i will try to download the zip and use it :D

Saiyato, 2018-03-12:
Yeah that jump should go away if you download the latest.

Hyldig01, 2018-03-12:
i have downloaded the newest, but unfortunately i can't install it from the webUI (latest version of volume) i have tried over SSH but still no luck, So i probably have to wait for volumio to have your newest version :D

Saiyato, 2018-03-12:
I've updated the readme, it's super simple to install now, the lazy way (lazy implying you haven't read the plugin documentation by Volumio ;) ).

Hyldig01, 2018-03-12:
Thank you so much :D 

just a small problem now... 
after installing the new version... i can't change anything in the plugin settings. 
I change the settings as the old version, press save, and the dropdown says "Saved settings" like normal. but however nothing is saved 

After a little frustration i tried to change the files in 
volumio@pod1:~/volumio-rotary-encoder-plugin/options$ 
and 
volumio@pod1:~/volumio-rotary-encoder-plugin/config.json
but still nothing is saved nor working

Saiyato, 2018-03-12:
You're right, just fixed this, you only need to execute the script again (it will download a new zip)

Hyldig01, 2018-03-12:
Thank you so much :D 

everything works great now :D

Saiyato, 2018-06-11:
Awesome, good to know, I've noticed that some actions don't work, I'll try patch them asap.

Cheers

closed 2018-06-11 by Saiyato