Saiyato/volumio-rotary-encoder-plugin #5: No sound after installing

hailogugo, 2018-04-02:
Hi,
I use a Hifi Amp which will install like a HifiBerry amp.
When I use it without the rotary it work fine.
If i use it with rotary after installing i do not have any sound avaiable. Do anybody have an idea where it come from?

Thank you.

Hifi Amp
http://www.inno-maker.com/shop/rpi-hifi-amp/

Rotary is 
KY040

Software
Volumio 2.389

pkozlovskiy, 2018-05-15:
I have the same problem( on http://www.suptronics.com/miniPCkits/x5000.html
No sound after connect to gpio.

psasin, 2018-06-10:
I've got same issue. Using Hifi Amp and KY-040 same configuration as @hailogugo .
If I configure 1st encoder for volume control, then it is working fine, but as soon as I configure 2nd encoder for playback control i have no sound. I need to disable 2nd encoder and restart volumio.
Configuration of first encoder (working):

- CLK pin = 23

- DT pin = 25
- SW pin = 23

-

Saiyato, 2018-06-11:
Ok, for my understanding, the problem occurs only when you install a second encoder?
Also, have you configured the same pin for CLK and SW on the first encoder?

Also, the documentation of both the Suptronics and Hifiberry contain detailed descriptions of the GPIO pins you can't use, 19 being one of them ;) Pins 18-21 are normally used for the sound interface, so you can't use them for anything else if you want to boards to work.

HifiBerry: https://www.hifiberry.com/build/documentation/gpio-usage-of-hifiberry-boards/
Suptronics: http://www.suptronics.com/miniPCkits/x5000.html

I trust this will solve your problems, if not I can re-open the issue. 

@psasin this means your second encoder can't use pin 19 for CLK.

closed 2018-06-11 by Saiyato