Saiyato/volumio-rotary-encoder-plugin #10: zou deze plugin moeten werken ?

marcelveldt, 2018-07-31:
Hoi, fantastisch werk met je addons!

Ik probeer vandaag op 2 vers geïnstalleerde Volumio players de rotary plugin aan de praat te krijgen  maar zonder resultaat. Ik kan de waardes opgeven in de configuratie maar verder dan dat komt het niet. De rotary encoder reageert niet op commando's.

Ik gebruik BCM pinnen 17, 27 en 22 voor de rotary encoder module.
Met een python script werkt het wel dus de connectie is OK.

Is er ergens logbestand ofzo wat ik kan raadplegen ?

Saiyato, 2018-07-31:
Hoi Marcel,

Bedankt! Ik neem aan dat de rotary encoder ook voorzien is van 3v of 5v, dus dat deel sla ik even over.
Er wordt inderdaad gebruik gemaakt van de BCM/GPIO nummers, dus dat zou ook goed moeten gaan.

In de instellingen is het mogelijk 'debug logging' te activeren, deze wordt op de console getoond. Dus met `sudo journalctl -f` is deze te volgen.

Het is een vrij basic aansturing met via de onoff module, dus erg veel aanvullende logging heb ik helaas niet. De debug logging laat bij 'no action' in ieder geval zien of er een click of tick gedetecteerd is. Wellicht dat dat helpt, anders hebben we misschien een bugje te pakken, dat kan altijd natuurlijk.

Ik hoop dat dit helpt.

marcelveldt, 2018-07-31:
Bedankt voor je reactie!
Volumio lijkt in een loop te zitten met opstarten nu de plugin actief is. In de logging zie ik hem telkens opnieuw opstarten.

Ik spotte trouwens deze error:

Jul 31 19:02:06 volumiostreamer volumio[4221]: /data/plugins/user_interface/rotaryencoder/index.js:270
Jul 31 19:02:06 volumiostreamer volumio[4221]: self.firstEncoder.on('rotation', direction => {
Jul 31 19:02:06 volumiostreamer volumio[4221]: ^
Jul 31 19:02:06 volumiostreamer volumio[4221]: TypeError: Cannot read property 'on' of undefined
Jul 31 19:02:06 volumiostreamer volumio[4221]: at rotaryencoder.constructFirstEncoder (/data/plugins/user_interface/rotaryencoder/index.js:270:20)
Jul 31 19:02:06 volumiostreamer volumio[4221]: at rotaryencoder.onStart (/data/plugins/user_interface/rotaryencoder/index.js:39:8)
Jul 31 19:02:06 volumiostreamer volumio[4221]: at PluginManager.startCorePlugin (/volumio/app/pluginmanager.js:497:30)
Jul 31 19:02:06 volumiostreamer volumio[4221]: at HashMap.<anonymous> (/volumio/app/pluginmanager.js:609:29)
Jul 31 19:02:06 volumiostreamer volumio[4221]: at HashMap.forEach (/volumio/node_modules/hashmap/hashmap.js:172:10)
Jul 31 19:02:06 volumiostreamer volumio[4221]: at HashMap.proto.(anonymous function) [as forEach] (/volumio/node_modules/hashmap/hashmap.js:184:7)
Jul 31 19:02:06 volumiostreamer volumio[4221]: at PluginManager.startCorePlugins (/volumio/app/pluginmanager.js:608:19)
Jul 31 19:02:06 volumiostreamer volumio[4221]: at PluginManager.startPlugins (/volumio/app/pluginmanager.js:77:10)
Jul 31 19:02:06 volumiostreamer volumio[4221]: at new CoreCommandRouter (/volumio/app/index.js:62:24)
Jul 31 19:02:06 volumiostreamer volumio[4221]: at Object.<anonymous> (/volumio/index.js:35:21)
Jul 31 19:02:06 volumiostreamer volumio[4221]: at Module._compile (module.js:652:30)
Jul 31 19:02:06 volumiostreamer volumio[4221]: at Object.Module._extensions..js (module.js:663:10)
Jul 31 19:02:06 volumiostreamer volumio[4221]: at Module.load (module.js:565:32)
Jul 31 19:02:06 volumiostreamer volumio[4221]: at tryModuleLoad (module.js:505:12)
Jul 31 19:02:06 volumiostreamer volumio[4221]: at Function.Module._load (module.js:497:3)
Jul 31 19:02:06 volumiostreamer volumio[4221]: at Function.Module.runMain (module.js:693:10)

marcelveldt, 2018-07-31:
ok, ben een stapje verder...
ik gebruik consequent pins 17, 27 en 22 voor de rotary encoder. 
hiervoor had ik dietpi met een python script draaien op deze spelers zonder enig probleem.
Nu lijkt het erop dat pin 17 niet te gebruiken is. Zodra ik die opgeef gaat de hele handel weer in bootloop.

Ook ff snel het python script verhuisd naar volumio als test... ook daar: failed to add edge detection op BCM 17.

Dus 2 dingen: 

1) de plugin crasht als er iets mis gaat met toegang tot de GPIO en daardoor komt volumio in een bootloop.

2) BCM pin 17 is om de 1 of andere vage reden niet te gebruiken.

marcelveldt, 2018-07-31:
ok, gevalletje RTFM dit... diep verstopt in de handleiding van de Allo DigiOne (de geluidskaart die ik via i2s aanstuur) vermeld dat BCM17 gebruikt wordt. Vaag dat het voorheen met dietpi wel werkte maargoed...

de plugin doet het nu!

kleine puntjes van kritiek:

- ook al gebruik ik een pullup, de gevoeligheid is best extreem. denk dat er ergens in de code nog een debounce toegevoegd moet worden.

- het feit dat het script de hele volumio laat crashen is wel beangstigend :-)

Saiyato, 2018-08-04:
Dat hij Volumio crasht is inderdaad niet netjes, zal kijken of ik voldoende try-catches kan toevoegen en/of op de juiste plek. Daarom is het handig al mensen bugs melden ;)

Dat een GPIO al in gebruik is kan ik niet tonen, veel HATs gebruiken GPIO 17, ik denk dat dit aan de gebruiker is om te weten/uit te zoeken.