Saiyato/volumio-rotary-encoder-plugin #11: Documentation should explicitely mention need for pullup on encoder lines

wodz, 2018-08-03:
The code expects encoder lines to have pullups. This is only mentioned in source code. I see the code uses sysfs entries to interact with gpios so there is no way to force pullup from code. On rpi one can use device tree overlay to force pullups (or python/c api). External pullup is an option too.

Saiyato, 2018-08-04:
Hi,

I'm not sure I follow, pullup shouldn't be required, it just reads the new value, appends it to the last and uses that result to determine the direction.

Unless you're talking about the click function, this one does imply pullup indeed. I haven't had any trouble with it, so I assumed it automatically pulls up.

I can add it to the documentation of course, but I'm also trying to understand whether or not it's actually needed and what the default values are. :)

Can you help me with that?

wodz, 2018-08-13:
The hardware is Rpi3 with JustBoom amp HAT (pinout https://www.justboom.co/technical-guides/boards-pinout/). Rotary encoder (https://www.tme.eu/pl/Document/8c1b1e629e264c85ef5bd42048515675/PEC11R.pdf) is connected to GPIO23, GPIO24, GND and push switch to GPIO15. My understanding (supported by suggested circuit in linked PDF) is that input lines should be pulled up. Default configuration of GPIO23 and GPIO24 is definitely not pulled up (I suspect weak pulldown rather).

Saiyato, 2018-10-02:
I read up a bit on pull-ups, the encoder I used (KY-040, badly documented) actually has internal 10k pull-ups (when + is connected), this only pulls the encoder contacts up. A 0,47u capacitor should debounce the signals enough, see: http://forum.arduino.cc/index.php/topic,242356.0.html

This means, if you are using encoders without pull-up resistors, you indeed should add those. I'm starting to understand the electronics of it better now. Correct me if/when I'm wrong.

So yes, you are absolutely right, pull-up is implied and should be mentioned. Also, almost everywhere I go I read _external_ pull-up is advised; so I will advise that as well. The PEC11R is quite accurate and will require careful pulling up and limiting I presume. For both the encoder and the switch to be precise.

What I don't yet fully understand is when you would use pull-up and when pull-down. Apart from the code expecting either of course. Is there a valid reason to use pull-down instead? For example to minimize wear and tear?

Saiyato, 2018-10-02:
Also, the Bourns PEC11r encoder works on 10mA @ 5V, so the pull-up value would be 5k and for 3.3 that'd be 3.3k, right?

mundodisco8, 2019-01-19:
The pull-up resistor is only there to drive the voltage up when there is no connection (to prevent the pin from floating). Any current circulating through it is dissipated as heat and wasted. So using a low value will cause a high current to go through it. It's not a big deal, because the raspberry pi is plugged, and not battery operated, so 10mA more or less on something that draws up to 800mA is not going to change things much, is it?
The logical thing to think then, is that the higher the pull up resistor value, the less current it will draw, and less energy is wasted. The problem is that, if for any reason, any other resistor is connected in parallel, the parallel of a resistor R with another that is much higher (>10R) equals to more or less R, so your pull up isn't really doing much (it's what it's called a weak pull up).

If you are thinking "argh! you are boring! just tell me what to use!", then use 10k, don't overthink it. That's the standard value, that should work. And why 10? well, it's a good middle ground value (if you don't have a 10K around but have a 7K or a 15K or a 20K, use them, don't sweat it). Why not 5k, as you say in your previous message? because what you list of 10mA and 5V is the max current the switch can handle, you don't want to have that much going through it, and you don't need to change it to 3.3k when working with 3.3V

Also, keep in mind that if you are filtering with an RC filter, the value of the pull up might affect your R in the RC filter (it might be seen in series or in parallel to your R of the RC filter).

Hope my explanation is not too convoluted

wodz, 2019-02-18:
dtoverlay solution for those interested to use internal pullups
https://github.com/fivdi/onoff/wiki/Enabling-Pullup-and-Pulldown-Resistors-on-The-Raspberry-Pi

wodz, 2020-03-25:
even easier solution is to activate pullups using gpio directive in config.txt. I use such line for example:
gpio=22-24=ip,pu

mundodisco8, 2020-03-25:
My advice would be to use external pull ups. At the end of the day, you are
already adding hardware (the rotary encoder), so adding an extra resistor
is trivial. With external pull ups you can choose the pull up value and
control rise times, which is critical in a rotary encoder, given that the
rising time determines "how fast" you can rotate it.
I agree that choosing the value can be daunting if you don't know enough
electronics, but again, as i said before, i would only choose internal is
there's no chance to add an external resistor.