Saiyato/volumio-rotary-encoder-plugin #13: Volume control does not work

ejmorgan55, 2019-01-16:
I have tried to get this plug in to function for hours; no joy. I have set it to clk 23; dt 24; switch 17. It simply does nothing at all. I have installed the plug in from volumio; enabled; saved the results; restarted. Nothing. I have tried other pins: 14; 15 4. still nothing. Is there a problem with the plug in? I am using rotary encoders from ADAfruit; they have worked well on other builds with other systems. Any help would be appreciated.

Saiyato, 2019-01-17:
Hi,

Which version of Volumio, the plugin and what hardware are you using? 
Also, can you enable debug logging and post the logging (make sure all personal information - if any - is stripped beforehand)?

ejmorgan55, 2019-01-18:
Any comment on the encoder not working i.e. has something changed with the new version of Volumio? Also; how do I get to the log for the plug in?

Saiyato, 2019-01-19:
Which version of the plugin are you using? That's very important, because there are some fixes here and there and at this point I'm uncertain if all fixes have been committed just yet.

Also the plugin logs to the standard Volumio logfile, you can also tail the journal live to see what's happening.

ejmorgan55, 2019-01-19:
Version 1.0.6 as presented on the Volumio plugin GUI. I believe this is the current version. I enabled the log via the sudo journalctl -f function. I seen real time logging for all functions; except for the encoders. No logging is produced for any coder function. I have one encoder set up on 4; 14; and 15 GPIO pins, for volume and mute control. The log indicates no input; as a consequence, I would suggest the code is not functioning at all in the new version of volumio. Thanks.

ejmorgan55, 2019-01-19:
The encoder is now working intermittently i.e. the mute/unmute function works on the switch; however, still no volume control; albeit, it is now logging when moved:

Jan 19 19:03:59 volumio volumio[879]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Jan 19 19:03:59 volumio volumio[879]: info:
Jan 19 19:03:59 volumio volumio[879]: ---------------------------- Client requests Volume -
Jan 19 19:03:59 volumio volumio[879]: info: VolumeController::SetAlsaVolume-
Jan 19 19:03:59 volumio sudo[5713]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/sbin/ifconfig wlan0
Jan 19 19:03:59 volumio sudo[5713]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 19 19:03:59 volumio sudo[5713]: pam_unix(sudo:session): session closed for user root
Jan 19 19:03:59 volumio volumio[879]: info: CoreStateMachine::pushState
Jan 19 19:03:59 volumio volumio[879]: info: CoreStateMachine::getState
Jan 19 19:03:59 volumio volumio[879]: info: CoreCommandRouter::volumioPushState
Jan 19 19:03:59 volumio volumio[879]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Jan 19 19:03:59 volumio sudo[5725]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/sbin/ifconfig wlan0
Jan 19 19:03:59 volumio sudo[5725]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 19 19:03:59 volumio sudo[5725]: pam_unix(sudo:session): session closed for user root
 
Not sure if you can make any sense of all this.

Saiyato, 2019-01-20:
Hi, 

That version is old, you need to either manually update the files from this repo, or wait till the PR for the new version gets approved by the Volumio team.

If you update manually you can verify that the plugin is working with the new (and very improved) code.

ejmorgan55, 2019-01-20:
I tried to update with your GitHub code:

volumio@volumio:~/volumio-rotary-encoder-plugin$ git pull origin master
From https://github.com/Saiyato/volumio-rotary-encoder-plugin
 * branch            master     -> FETCH_HEAD
Already up-to-date.
volumio@volumio:~/volumio-rotary-encoder-plugin$

As above it show up to date? It appears the code on GitHub and the code on Volumio are the same?

Is there an update I am missing?

Saiyato, 2019-01-21:
Yes I think you are pulling from the Volumio repo, you my personal repo to get the latest version.

ejmorgan55, 2019-01-21:
I am pulling from Github: From https://github.com/Saiyato/volumio-rotary-encoder-plugin. Is there another location? 

Also; I was able to get the volume control working by manually turning on the pull resister in the RP. Menu control will not function at all--no output regardless of the settings. 

Let me know if I am pulling from the wrong place.

Saiyato, 2019-01-22:
If you pulled from here and installed the version (this is imperative), your version should read 1.2.2
You might want to (re)install the needed node modules, because if you're pulling, you're subject to major changes.

Saiyato, 2019-05-17:
The plugin is now available in the store, no need to manually pull from the repo, the version in the store is up-to-date. I will update this repo when I have the chance to play a bit more to make it more snappy.

closed 2019-05-17 by Saiyato