Saiyato/volumio-rotary-encoder-plugin #14: Will not install

zhudnall, 2019-03-05:
So I'm running 2.55, since the plugin available from the GUI is 1.0.6 and doesn't seem to planned for upgrade...

I cloned the repo and attempted to run the install with su, says it's installed, checked the install location, it's a "no go".

I've tried with every volumio distro since last summer, no dice. What's going on?

NeillRG, 2019-03-21:
I also am having trouble getting the plugin installed. I tried the lazy install fork but (after following the instruction) it borked my Volumio install, (lost WIFI on my PI0) and I had to reformat my SD. Have you had any joy?

Jest0r, 2019-03-24:
You want to unzip volumio-rotary-encoder-plugin.zip form the repo, cd into there and run the 'volumio plugin install' from there. worked for me :)

Saiyato, 2019-05-17:
Hi there,

The plugin has been incorporated in the store, no need to manually install anymore. ;)

closed 2019-05-17 by Saiyato