Saiyato/volumio-rotary-encoder-plugin #16: Rotary encoder does not work

calimero22, 2019-05-08:
Dears,

many thanks for this great plugin although I didn't manage to work so far. I am a real newbie to this. So, probably sorry for this question. I use following hardware/software setup:
- raspberry pi 3 Model B+
- iqaudio DAC+
- rotary encoder KY-040
- volumio Version 2.575

I connected the rotary encoder to following pins:
- CLK: pin 18
- DT: pin 16
- Ground: pin 14

I entered the following inputs to the plugin settings:
- CLK Pin Number: 18
- DT Pin Number: 16
- SW (Switsch) Pin Number: 0
- Choose Coding: Default
- Detent Action: Controle volume
- Button Action: No action
- Long Press Action: No action

Could you please advice me what I have to change that this rotary encoder will work? Many thanks in advance for any help and support.

Saiyato, 2019-05-09:
Hi there,

Can you enable debug logging and trail the journal entries? You can do so with the following command:
`sudo journalctl -f`

Please strip any IP-addresses and other sensitive information before posting.

Steps to perform:

1. Enable debug logging
2. Set up the trail
3. Use the encoder

I expect to see the plugin to at least log what's happening, that should indicate whether or not the encoder is working as intended. If not, there are two options:

1. Check the wiring, the pin number is not the GPIO number (e.g. GPIO 18 is pin 12)
2. Test the encoder itself

calimero22, 2019-05-09:
Dear @Saiyato 

Many thanks for your prompt reply. I have checked the setup with different rotary encoders but without success. I have checked the wiring. Please see the [picture](https://imgur.com/8vGqDB5)
- CLK: Pin 18 (green cable)
- DT: Pin 16 (red cable)
- Ground: Pin 14 (black cable)

Please find below the output for the command `sudo journalctl -f`:

`volumio@volumio:~$ sudo journalctl -f
-- Logs begin at Thu 2019-05-09 08:11:53 UTC. --
May 09 08:29:13 volumio systemd[7885]: Starting Basic System.
May 09 08:29:13 volumio systemd[7885]: Reached target Basic System.
May 09 08:29:13 volumio systemd[7885]: Starting Default.
May 09 08:29:13 volumio systemd[7885]: Reached target Default.
May 09 08:29:13 volumio systemd[7885]: Startup finished in 41ms.
May 09 08:29:13 volumio systemd[1]: Started User Manager for UID 1000.
May 09 08:29:13 volumio sshd[7858]: lastlog_openseek: Couldn't stat /var/log/lastlog: No such file or directory
May 09 08:29:13 volumio sshd[7858]: lastlog_openseek: Couldn't stat /var/log/lastlog: No such file or directory
May 09 08:29:25 volumio sudo[7975]: volumio : TTY=pts/0 ; PWD=/home/volumio ; USER=root ; COMMAND=/bin/journalctl -f
May 09 08:29:25 volumio sudo[7975]: pam_unix(sudo:session): session opened for user root by volumio(uid=0)
May 09 08:29:36 volumio kernel: usb 1-1.3: USB disconnect, device number 22
May 09 08:29:37 volumio kernel: usb 1-1.3: new low-speed USB device number 23 using dwc_otg
May 09 08:29:37 volumio kernel: usb 1-1.3: New USB device found, idVendor=045e, idProduct=00cb
May 09 08:29:37 volumio kernel: usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
May 09 08:29:37 volumio kernel: usb 1-1.3: Product: Microsoft USB Optical Mouse
May 09 08:29:37 volumio kernel: usb 1-1.3: Manufacturer: PixArt
May 09 08:29:37 volumio kernel: input: PixArt Microsoft USB Optical Mouse as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/0003:045E:00CB.0014/input/input19
May 09 08:29:37 volumio kernel: hid-generic 0003:045E:00CB.0014: input,hidraw2: USB HID v1.11 Mouse [PixArt Microsoft USB Optical Mouse] on usb-3f980000.usb-1.3/input0`

Many thanks in advance for your help!

Saiyato, 2019-05-09:
Hi there,

I also connect +5V to my rotary, not sure if it's really necessary, but I always do so.
Also check here: http://www.ardiotech.com/en/connect-ky-040-rotary-encoder-raspberry-pi/

I've seen schema's without +5V, but I never bother to check when or not it's needed ;)

calimero22, 2019-05-09:
Hi @Saiyato Saiyato,

thanks a lot for your hint for connecting +5V. Unfortunately, no success yet. I tried +5V and +3.3V.
I should receive a different type of rotary encoder (https://www.adafruit.com/product/377) tomorrow. I will try again and let you know. Many thanks for your support so far.

Do you have any hints for the other type of rotary encoder mentioned above? Do I need to consider something? I will plug in the 3 connections on the one side to the following pins as mentioned in this [tutorial](http://www.hagensieker.com/rotaryenc/index.php):
- left connector to pin 16
- middle connector to pin 14
- right connector to pin 18

Best regards

Saiyato, 2019-05-13:
Hi there,

I was again triggered by your remarks, as I'm pretty sure the plugin should at least do _something_. So I reviewed the photo you uploaded and you connected pins 14, 16 and 18. None of which are GPIO 16 and 18. Pin 14 = GND, pin 16 = GEN4 (or GPIO23) and pin 18 = GEN5 (or GPIO24).

Sources: 
https://imgur.com/8vGqDB5
https://cdn.sparkfun.com/assets/learn_tutorials/4/2/4/header_pinout.jpg
http://www.iqaudio.com/downloads/IQaudIO.pdf

So you need to use GPIO23 for DT and GPIO24 for CLK and probably still +5V as well. I'm guessing you just misinterpreted the pin numbers, as outlined in my first comment ;)

calimero22, 2019-05-13:
Dear @Saiyato 

Many thanks for your reply. Now, I am really confused. Do I need to put in 23 and 24 for GPIO23 and GPIO24 into the UI of your plugin instead of 16 and 18? Or do I need to switch the connectors? Are the connected pin 16 and 18 not GPIO23 and GPIO24? Did I connect the wrong pins? Many thanks for your help and sorry to bother you with this newbie stuff. I really appreciate your help @Saiyato !

Saiyato, 2019-05-13:
Hi @calimero22,

You have connected GPIO23 and GPIO24, but they have (header) pin numbers 16 and 18. The plugin expects GPIO numbers, not pin numbers on the header. Just configure DT = 23 and CLK = 24; see if that yields the results you expect ;) Oh and do connect the +5V pin as well, just to be sure

You did not connect the wrong pins, you just need to configure them using the GPIO numbers instead.

calimero22, 2019-05-14:
Dear @Saiyato 

I have tried several things based on your hint but without success. I have (please see the pictures)
- replaced the microSD card
- pinned the cables to DT = GPIO 23 (Pin 14, white cable) and CLK = GPIO 24 (Pin 16, green cable) as well as Ground (Pin 12, black cable) ([picture](https://imgur.com/Oe40pEQ))
- changed the typ of rotary encoder to the one you recommend
- freshly reinstalled Volumio
- installed your plugin
- put in 24 and 23 to the setting of your plugin ([picture](https://imgur.com/7OKsL1c))
- tested it with iqaudio DAC+ and Hifiberry DAC+
**-sound works well**
**- no success to get run the rotary encoder**

I have setup another SD Card with moodeaudio and configured the same rotary encoder. This worked straightaway. Do you have any idea what I should try next? Thanks in advance.

Best regards

calimero22, 2019-05-14:
Additionally, here is the journal
`volumio@volumio:~$ sudo journalctl -f
-- Logs begin at Tue 2019-05-14 21:42:59 UTC. --
May 14 22:16:46 volumio systemd[7425]: Starting Basic System.
May 14 22:16:46 volumio systemd[7425]: Reached target Basic System.
May 14 22:16:46 volumio systemd[7425]: Starting Default.
May 14 22:16:46 volumio systemd[7425]: Reached target Default.
May 14 22:16:46 volumio systemd[7425]: Startup finished in 27ms.
May 14 22:16:46 volumio systemd[1]: Started User Manager for UID 1000.
May 14 22:16:46 volumio sshd[7398]: lastlog_openseek: Couldn't stat /var/log/lastlog: No such file or directory
May 14 22:16:46 volumio sshd[7398]: lastlog_openseek: Couldn't stat /var/log/lastlog: No such file or directory
May 14 22:16:50 volumio sudo[7466]: volumio : TTY=pts/0 ; PWD=/home/volumio ; USER=root ; COMMAND=/bin/journalctl -f
May 14 22:16:50 volumio sudo[7466]: pam_unix(sudo:session): session opened for user root by volumio(uid=0)`

calimero22, 2019-05-16:
Hi @Saiyato 

Finally, I managed it! I pinned the KY 040 rotary encoder to the following GPIO/pin as outlined in this [tutorial](https://www.stuffaboutcode.com/2015/05/raspberry-pi-and-ky040-rotary-encoder.html):
- CLK to GPIO 5
- DT to GPIO 6
- SW to GPIO 13
- Ground to pin 6
- 3.3V to pin 1

Is there any possibility to adjust the increasing/decreasing steps of the rotary encoder? Currently, the volume increases/decreases by 10 or 11. Is there any possibility to adjust the script accordingly? Many thanks in advance for your hint.

Best regards,
Calimero22

Saiyato, 2019-05-17:
Hi @calimero22 , good to hear to managed to get it working. Weird that MoOde would expose different behavior, it might be in the underlying OS version and/or settings or drivers installed on the image (Volumio might have different hardware drivers because of a different relationship to suppliers). However, that's been covered, so no need to investigate that.

As for the detent action, I'm just sending 'volume up', which should result in a single volume up command; i.e. +1. If you're turning the knob more franticly this number might increase to unpredictable values. This is caused by a delay I'm not really able to solve, somewhere between the hardware and Volumio (might even be in de NodeJS part). The only thing you can do is either have it talk to ALSA directly or buffer the detents, both of which require quite some rewriting of the code.

I will close this issue, because the subject 'rotary encoder does not work' implies that it actually doesn't work, which it does ;) You can still comment however, it just like to run a tight ship when it comes to information.

closed 2019-05-17 by Saiyato