Saiyato/volumio-rotary-encoder-plugin #18: Rotary encoder plugin not storing configuration after reset

dubaleeiro, 2020-03-21:
Hello,

for some reason I can't understand, the rotary encoder configuration (pins, detent action, etc) is "lost" after rebooting the Pi and only gets back to work when the configuration is saved in the UI.

According to the log information the encoder config is been loaded correctly, so I really don't understand what could be wrong...:

`Mar 21 14:03:06 volumio volumio[895]: info: [Rotary encoder] Loaded configuration: {"longPressThresholdInMilliseconds":{"type":"int","value":1000},"first_encoder_CLK":{"type":"int","value":24},"first_encoder_DT":{"type":"int","value":23},"first_encoder_SW":{"type":"int","value":0},"first_encoder_encoding":{"type":"int","value":0},"first_encoder_detentActionType":{"type":"int","value":1},"first_encoder_buttonActionType":{"type":"int","value":0},"first_encoder_longPressActionType":{"type":"int","value":0},"second_encoder_CLK":{"type":"int","value":8},"second_encoder_DT":{"type":"int","value":25},"second_encoder_SW":{"type":"int","value":0},"second_encoder_encoding":{"type":"int","value":0},"second_encoder_detentActionType":{"type":"int","value":2},"second_encoder_buttonActionType":{"type":"int","value":0},"second_encoder_longPressActionType":{"type":"int","value":0},"enable_debug_logging":{"type":"boolean","value":true}}
`

Can you help me on that? Thank you!

Raspberry Pi Zero W V1.1
Innomaker HiFi AMP Hat
System Version: 2.714
Plugin Rotary Encoder 1.2.2 (maybe obvious information after the one above...)
Encoder #1: CLK=24; DT=23;detent action=1 (volume control)
Encoder #2: CLK=8; DT=25;detent action=2 (prev/next)

Saiyato, 2020-03-25:
I will have to investigate, sounds like an initiation problem, as the log does prove the settings are still there.

Does this happen on every boot? I.e. restart and cold boot?

dubaleeiro, 2020-03-25:
Hi, thanks a lot for your reply. Yes, that happens on every boot (cold boot and restart). Only when I save again the pin configuration through UI interface the rotary encoder works.

This plugin is great, so it is a pity that it is not working... (for me at least?).

I can send you the full initialization log if you want. 

Thanks!

fabio323ti, 2020-04-14:
also have same issue, at reboot should "save" settings to work, did you solved?

Saiyato, 2020-04-14:
Hi @fabio323ti what's your setup? Version number etc? 
My test-setup keeps its config

fabio323ti, 2020-04-14:
volumio2 (latest release) and plugin (installed from volumio webUi) rotary encoder version 1.2.2

Saiyato, 2020-04-14:
Thanks, are you running on a Pi(0, 1, 2, 3, 3+ or 4), x86 or other architecture?

fabio323ti, 2020-04-14:
sorry misundestand , Pi 3b+ , x86

dubaleeiro, 2020-04-14:
It would be great to have a solution here... I am still stuck and can’t solve it by myself...

fabio323ti, 2020-04-14:
its just settings lost on reboot... i guess with some small script will solve...

dubaleeiro, 2020-04-23:
@Saiyato , may I ask whether you had some time to take a look at this issue? Thanks!

Saiyato, 2020-04-23:
I think I might have a clue, do you have the switch enabled on your encoder? Or is that pin value set to zero (0)? I could easily patch, and I think I actually should... setting that value to anything but 0 should then fix your problem (e.g. 100 which doesn't exist on the Pi).

closed 2020-04-23 by Saiyato

dubaleeiro, 2020-05-03:
I have just tested again and unfortunately it did not work. Volumio UI does not initialize when the value for switch is different from „0“. Very strange...

Saiyato, 2020-05-07:
Can you post some logging, I have no clue as to what your settings are :)

Saiyato, 2020-05-07:
Never mind, I think I found the culprit... I still had to update the zipfile. You were probably still using version 1.2.2. The new commit should take care of this, v1.2.4 (please check the version before testing).

dubaleeiro, 2020-05-09:
Hi, it is working now with version 1.2.4, thank you !! Just the switch function is not working yet - is that what you also expected?
The new commit has apparently not worked either - I had to install v1.2.4 manually.
Cheers,
Eduardo

Saiyato, 2020-05-11:
Hi Eduardo,

Good to hear! As for the switch; you configured 0 (according to your first post), which means no switch.

The new commit still has to be accepted by Volumio, this is not something I can do from this repo. Until that time you will need to install manually unfortunately.

dubaleeiro, 2020-05-11:
Hi @Saiyato ,
I have also configured a normal pin for the switch (GPIO 16) and it did not work. Should it have worked? I am suspecting there something wrong with my Keyes encoder...

Saiyato, 2020-05-11:
I think this is a different problem than not storing the configuration. 

If you enable debug logging, you can see if the button press is detected (same as detents). This should help you in ascertaining whether the encoder is faulty, or we've stumbled on a new bug ;)

dubaleeiro, 2020-05-17:
Hi @Saiyato , I changed the encoder and the switch is working as expected. Thank you very much !

Right now I am only facing the issue reported in #19

Saiyato, 2020-06-04:
I see this one is also raised at the Volumio repo volumio/Volumio2#1977

commit 8a112b64 (2020-04-23): Patched not initializing encoders on boot

There was a wrong check in whether the encoders should be initialized on boot, this fixes #18