Saiyato/volumio-rotary-encoder-plugin #19: Volume control returning NaN

spong-e, 2020-05-15:
Hi @Saiyato, having a strange issue where I have connected a rotary encoder with mixed success. If I select 'Control Volume' as the Detent action, then the volume doesn't change and the display updates to `NaN/100` as shown in the screenshot.

[image]

Im confident the encoder is working as if I select a different action, eg, Media Control (prev/next) this works fine, so that suggests the encoder is ok and that it is wired up ok. Just for clarification m using the following pins...

CLK - GPIO17 (pin11)
DT - GPIO18 (pin12)
+ - 3v3 (pin1)
GND - GND (pin6)

 I have a second encoder attached, which is also working ok (IM using that one to control next/prev, toggle play/pause) So its only the volume control I am having an issue with.

Saiyato, 2020-05-15:
Hi,

Just a few questions:
1. Are you using a DAC?
2. Can you control the volume fine with the WUI (web user interface)?
3. Can you hear anything when the volume displays this NaN status?
4. Does this also happen when you execute the command "volumio volume plus"?

I can only reproduce when I send the command "volumio volume up", but that command should not be in the plugin anymore.

If the command doesn't give the same result as turning the knob, please add some logging

spong-e, 2020-05-15:
Hi,

Thank you for such a fast reply!

1. No am using headphone jack.
2. Yes UI works as expected.
3. When get the NaN value, the volume stays the same.
4. Using the command "volumio volume plus" worked.

The other thing I should add is that the encoder works occasionally, and can start/stop working without a reboot/config change/anything changing.

So annoyingly its currently working, I can usually get it back working by changing the action to another option, eg, next/prev and then back to volume.

Saiyato, 2020-05-15:
Hi,
This sounds like a HW issue, but lets not jump to conclusions ;) If you can just enable debug logging and attach some logging of when it fails. That'd be very helpful.

Also, after reboot initialization has been fixed in v1.2.4, it should be in the plugin store soon. Or you'll have to install manually.

spong-e, 2020-05-15:
I have a feeling its the initialization issue, as have turned it off and on a few times now and Volume control wont work until I change to another action and then back to Volume control.

I can wait until v1.2.4 thank you 👍 

On a totally unrelated note, do you know when your PR for the LCD plugin is going to be merged? :-)

dubaleeiro, 2020-05-17:
Hello all,
unfortunately I can confirm this issue also with Rotary Encoder v1.2.4 and Volumio 2.773. Here is the correspondent LOG:

`May 17 11:54:41 volumio volumio[952]: info: [Rotary encoder] Encoder #1 rotated left
May 17 11:54:41 volumio volumio[952]: info: [Rotary encoder] Executing command: /usr/local/bin/volumio volume minus
May 17 11:54:41 volumio volumio[952]: info:
May 17 11:54:41 volumio volumio[952]: ---------------------------- Client requests Volume -
May 17 11:54:41 volumio volumio[952]: info: VolumeController::SetAlsaVolume-
May 17 11:54:41 volumio volumio[952]: error: Cannot get ALSA Volume: Error: Alsa Mixer Error: failed to parse output
May 17 11:54:41 volumio volumio[952]: info: CoreStateMachine::pushState
May 17 11:54:41 volumio volumio[952]: info: CoreStateMachine::getState
May 17 11:54:41 volumio volumio[952]: info: CoreCommandRouter::volumioPushState
May 17 11:54:41 volumio volumio[952]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
`

Saiyato, 2020-05-20:
Hi @dubaleeiro , it looks like a problem that exists outside of the plugin, as you can see in the log, the plugin calls the function volumio volume minus:

`May 17 11:54:41 volumio volumio[952]: info: [Rotary encoder] Executing command: /usr/local/bin/volumio volume minus `

Does the same occur if you call that command from the CLI?

dubaleeiro, 2020-05-21:
Hello @Saiyato , I also believe now the problem is not with the plugin. I have another Pi running Volumio 2.729 and the volume control via encoder is working just fine.

To you question, it does not work via CLI (when I trigger `/usr/local/bin/volumio volume minus` for example). Just another comment: when the volume is displayed in the WUI as NaN just some "crack" noise can he heard from the speakers.

Any further hints, please?

Dave0616, 2020-05-23:
Im having this issue as well but im running a DAC, without the dac I can get it working fine, and with the dac I can get the encoder to prev/next, but if I try to control the vol it gets a NaN or lets me change the volume by 1 notch (if I have it set to change the volume by 5 I rotate the encoder it maxes out too 100 but i can go from 95 to 100) any ideas on how to get around this?

Saiyato, 2020-05-23:
Hi @Dave0616 ,

This adds to the suspicion it’s a mixer issue, rather than a plugin issue (the plugin just uses standard Volumio commands). What DAC are you using? Maybe I have a similar DAC for debugging purposes.

Dave0616, 2020-05-23:
this is the one im using
https://www.amazon.com/gp/product/B07D13QWV9/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

I just tried 2.413 and its working much better, im not sure yet if its the oled im trying to install as well or just the version of volumio yet

Dave0616, 2020-05-23:
I just reattached the oled screen and went through for the install for that, its all working , having volume control on hardware has issues (if I go below 31 it wont go above 31 after that) and I did that in the opposite order than the newest version of volumio (I installed the oled soft first then rotary encoder on that version) and it took me 2 tries, not sure what happened the first time but after enabling the rotary plugin the pi wouldent boot all the way up.

Saiyato, 2020-05-23:
Are you 100% sure you’re not using GPIO’s that are being used by the DAC? The DAC needs certain pins for functionality and audio, this might cause issues

Dave0616, 2020-05-23:
I cant be certain but I have tried pins all over at first, all have the same effect