Saiyato/volumio-rotary-encoder-plugin #21: "SW" on rotary encoder not working for short/long press

Sirnut, 2020-11-19:
I've set this plugin up and have been trying to get it to work with my Raspi 2 B+ IQAudio DAC+ and KY040 encoder

I'm using GPIO spots CLK:5   DW:6   SW:13

Knob rotation is registered with no issue, but I seem to have trouble getting the push function of the knob to be registered

Saiyato, 2020-11-19:
Hi @Sirnut ,

I did some googling, not sure why you picked GPIO13, but that pin has PWM functionality and is pulled down instead of up. See: https://elinux.org/RPi_BCM2835_GPIOs 

This means you have a few options, the list is not exhaustive, but these come to mind:

1. Use another pin, which is pulled up (see: https://raspberryautomation.com/raspberry-gpio-pins/)
2. Pull up the pin in config.txt `gpio=13=ip,pu` (documentation here: https://www.raspberrypi.org/documentation/configuration/config-txt/gpio.md)
2. Patch the code to suit your setup if you're unable to select another pin (e.g. due to hardware design, case, etc.)
3. Create a PR to dynamically select which pins are pulled which way per board revision (and x86). Just kidding, this is pretty much undoable :wink:

TL;DR
Pick another GPIO like 2, 3, 4, 7 or 8 (anyone pulled up), pull the GPIO up manually (in SW or HW) or patch the code to work with pulled-down pins.

closed 2020-11-19 by Saiyato

Sirnut, 2020-11-19:
Big thanks for the quick response. Full disclosure I've been out of the game for a while and completely forgot about the pull up/down status. I changed the pin in my config.txt and everything works as it should!

Thank you again for your help!!